roleoroleo/yi-hack-Allwinner-v2 #979: Yi Home 2K Pro not saving configuration

LuisDiazUgena, 2024-12-15:
Hi!

I just installed the custom fw on 3 cameras at home and one of those is not saving the configuration changes after a reboot.

<html>
<body>
<h5 class="tiny-margin strong" style="margin-top: 1rem; margin-bottom: 1rem; font-weight: 700; font-size: 2.4rem; line-height: 1.5; letter-spacing: -0.05rem; color: rgb(238, 238, 238); font-family: Arial, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(63, 63, 63); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">System</h5><hr class="no-margin" style="box-sizing: content-box; height: 0px; overflow: visible; margin: 0px; border-width: 1px 0px 0px; border-top-style: solid; border-top-color: rgb(225, 225, 225); color: rgb(238, 238, 238); font-family: Arial, sans-serif; font-size: 15px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(63, 63, 63); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">
Hostname | (none)
-- | --
Firmware Version | 0.3.3
Base Version | 12.0.51.01_202303091901
Model Suffix | y623
Serial Number | YFUSY94DA0HGAF231212
Local Time | Sun Dec 15 13:09:14 GMT 2024
Uptime | 0h 2m 12s
Load Average | 2.59 0.99 0.37
Free/Total Memory | 6268/60912 KB
Free SD Space | 100%

</body>
</html>

Please let me know what information should I provide in order to clarify this issue.

roleoroleo, 2024-12-18:
It could be a hardware issue or a power problem.
Please, try:

- another power adapter/cable
- another sd card

LuisDiazUgena, 2024-12-18:
Sorry I was super busy and forgot to update the repo. 

SD card was faulty and swaping to another SD card worked great.

closed 2024-12-18 by LuisDiazUgena