roleoroleo/yi-hack-Allwinner-v2 #982: r35gb Yi Camera cannot load the web interface

idanny76, 2024-12-16:
I am not sure what happened, but all of a sudden it stopped loading the web interface. I notice the SD card files were missing and I am guessing it could be due to me doing a SD Card Format. I added the files back to the SD Card, but its not installing the YiHack. I tried placing the previous YiHack firmware and it still did not work. How can I fix it?

idanny76, 2024-12-17:
Here is the firmware version it currently has. 
[image]

idanny76, 2024-12-17:
I was going to restore the backup in the SD card, but the steps are many and confusing. I was hoping flashing the older firmware or new firmware would bring the web interface back. The camera works fine, but no interface.

idanny76, 2024-12-17:
Problems solved.... for some reason is on Port 80 and not port 8080. I guess it I forgot since the rest of the Yi cameras are on 8080.

roleoroleo, 2024-12-18:
I recently changed the http port.
If you update the hack the port doesn't change.
If you reinstall the hack the port will change.

closed 2025-04-18 by github-actions[bot]