roleoroleo/yi-hack-Allwinner-v2 #985: yi-hack-Allwinner.gdrive folder structure

evider2002, 2024-12-25:
Hi,

I installed yi-hack-Allwinner.gdrive and it is working fine, but I have a question how to accomplish something specific.  

Google drive has a features called Google Drive versioning, that if you upload a file with the same name to the same folder, it will keep both files labeling them with a version number, instead of deleting the first version and keeping the second. Google will keep 100 versions before deleting the oldest one. I am trying to set up yi-hack-Allwinner.gdrive to upload one day of footage and then upload the next day to the same location where it will become a new version, instead of being uploaded to a new folder where it will not be merged as a new version to yesterdays file. 

The benefit of having gdrive upload this way is: 
A - I would have 100 days of storage on Google drive but it only uses up 1 days worth of my storage quota (only the most recent file version counts against storage.)
B - There is no need for file cleanup as google will automatically delete the oldest version when 100 versions are reached. 

The issue is that right now the setup is that Yi camera creates a new folder for each hour and labels it with year month day hour - 2024Y12M2**4**D23H, so then tomorrow the same hour folder will be labeled 2024Y12M2**5**D23H, therefore tomorrow's files will be going into a new folder. What I would like, is for the folder to just be labeled with the hour - 23H, so tomorrow's files will go into the same folder and be uploaded as a new version. 
In short, I would like for there to just be 24 folders corresponding to the hour, and then each day the same 24 folders will keep on being used again and again and again, but there will be 100 versions of each file for the past 100 days. 

How can I set this up? 
Thank you!

evider2002, 2024-12-30:
Thank you for your response [roleoroleo](https://github.com/roleoroleo) . 
I just opened the issue there, but the reason I put it here to begin with, is because there doesn't seem to be much activity going on there...

roleoroleo, 2025-01-01:
You can leave this issue open.

closed 2025-05-02 by github-actions[bot]