roleoroleo/yi-hack-Allwinner-v2 #987: h52ga - regular recording without motion detection

david-tub, 2024-12-26:
For some unknown reasons Im affected by regular small recordings without any motion detection. Almost exactly every 6 hours, small videos are recorded and uploaded to my FTP server. There is no motion at this time so I dont see any reason for starting the recording. The videos are about 40-60 seconds long. The recording frequency is: 07:33 - 13:33 - 19:33 - 01:33 ...

- h52ga
- cloud features disabled (private mode)
- recordings on motion detection are uploaded to a FTP server (I tried already to change the detection sentitivity)
- timelapse feature is enabled to take a photo once a day at 12 AM

Have I overlooked something or how could this be explained?
Thanks in advance!

roleoroleo, 2024-12-30:
Mmmmhhh...
This frequency is very strange.
Is there an external connection with other services? Like domotic server, web services...

david-tub, 2025-01-01:
The frequency is not "perfect" so Im not sure if there is a "real frequency" or just a "regular randomness" of recording.
However, there are no external services or other connections. Just the ftp-push is configured to upload all recordings to a FTP server. Of course, recording only on motion is enabled. Also, sound detection is disabled (my first assumption was that some random noise triggers the camera).

It is really strange.

roleoroleo, 2025-01-01:
I will test my h52ga, can you send me your configuration file?
Obscured...

david-tub, 2025-01-05:
[config_issue_987.zip](https://github.com/user-attachments/files/18311597/config_issue_987.zip)
Here is my complete config. I just changed the FTP password in the config file and re-zipped since GitHub does not allow to upload bz2 archives.

I continue testing out further configuration changes such as disabling snapshots or changing the Swap swappiness value. Thanks in advance for any ideas.

david-tub, 2025-01-08:
I have to correct myself. I have the **h51ga**. So sorry for that mistake.

However, I tried to change Swap swappiness value and force low resolution snapshot. No changes.
Next, I will disabled snapshot and swap file completely. Maybe this will have an impact.

roleoroleo, 2025-01-12:
My h51ga is not working at the moment (flash chip desoldered).
I need time to restore and test it.

closed 2025-05-14 by github-actions[bot]