roleoroleo/yi-hack-Allwinner-v2 #988: go2rtc + aac no audio on 0.3.3 y623

vekexasia, 2025-01-02:
As title says, it seems that

Standard + pcm => works
Alternative + pcm => only audio
Go2rtc + aac => only video

am i missing something?

vekexasia, 2025-01-03:
Did try different configuration parameters. seems audio is being passed to the stream but it's muted (using vlc)
highres >
[image]
lowres>
[image]

Using **ffmpeg** i can record and hear some chunks of audio... here chunks of ffmpeg log

```
[rtsp @ 0x5636e4987240] method SETUP failed: 461 Unsupported transport
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 3 with DTS 16, packet 4 with DTS 23404
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 7 with DTS 23411, packet 8 with DTS 45109
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 11 with DTS 45120, packet 12 with DTS 66631
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 15 with DTS 66649, packet 16 with DTS 91549
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 19 with DTS 91557, packet 20 with DTS 117098
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 23 with DTS 117113, packet 24 with DTS 139251
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 27 with DTS 139268, packet 28 with DTS 159595
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 31 with DTS 159612, packet 32 with DTS 181914
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 35 with DTS 181920, packet 36 with DTS 206323
[rtsp @ 0x5636e4987240] DTS discontinuity in stream 1: packet 39 with DTS 206331, packet 40 with DTS 231168
``

and a few of 
``` 
[aac @ 0x5636e49ac880] Queue input is backward in time
```

attached video recorded with ffmpeg where you can hear something but only in chunks.

https://github.com/user-attachments/assets/c9408e55-2180-496b-920a-3c96d2665257

I logged in via ssh to see the go2rtc config and it seems correct

```yaml
streams:
  ch0_0.h264:
    - exec:h264grabber -m y623  -r high#backchannel=0
    - exec:h264grabber -m y623 -r none -a#backchannel=0
  ch0_1.h264:
    - exec:h264grabber -m y623  -r low#backchannel=0

api:
  listen: ""

webrtc:
  listen: ""

rtsp:
  listen: ":554"
```

roleoroleo, 2025-01-03:
Yes, I know.
There is a warning in the release page: https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/tag/0.3.3
I'm not able to make it working, at the moment.
About alternative, I'll check it.

vekexasia, 2025-01-03:
TBh it works i can provide  a clip that works but has some issues ... if i upload it here the re-encoding seems to make it work. but when sharing the original file you'll notice the issue.

closed 2025-05-04 by github-actions[bot]