roleoroleo/yi-hack-Allwinner-v2 #990: What is the problem with all the links?

kumdzio, 2025-01-12:
I have strange issue with my y21ga (IFUSY) when connected to computer via usb cable it starts and works perfectly but when trying to connecti it to charger it is stuck and yellow/orange light.

So I assumed that it is software problem (something corrupted) but EVERY FUCKING link/file that has been shared via this repository or issues has been REMOVED. Why is that? Is it that problem to maintain the shared files or what?
@roleoroleo do you want people to pay you for help with the software or chinese manufacturers has tracked you down and requested to take the files down?

dreimer1986, 2025-01-12:
[image]

Where is your problem? You can expand the assets part, you know?

kumdzio, 2025-01-14:
Yes, i know. I meant the links in issues. There are couple places where people were needing help with their not working cameras and author @roleoroleo provided them some files to recover or flash into cameras to work. And afterwards those provided files has been removed from @roleoroleo 's comments. That is something that i do not understand. @roleoroleo can you please tell me why is that?

Also: sorry for being mad in first message I was little angry because the camera was not working and I couldn't find any solution.
For all those who have similar issue i have found a reason. It is because manufacturer of this camera does not care about any standards, especially the USB standard and he decided to hook up the RX and TX lines into D+ and D- pins in micro usb socket. So if you use standard usb cable there is possibility that you will pull-up or pull-down the RX/TX lines which is preventing camera from starting. So either use only 2-wire cable or modify your camera. I have cutted traces so RX and TX are no more connected to USB socked and now any USB cable will work with my camera.

roleoroleo, 2025-01-21:
> Yes, i know. I meant the links in issues. There are couple places where people were needing help with their not working cameras and author @roleoroleo provided them some files to recover or flash [...]

It depends on the file.
Often users share a backup that contains personal data.
mtd4, mtd6 and mtd7 may contain "secret data" like SSID, WPA key and serial number (there are many vulnerabilities on serial number).
So, I prefer to delete files if I am not sure about the content.
But I have never asked for money to help a user recover his cam. I can share a clean backup (if I have it) without any problem.

closed 2025-05-22 by github-actions[bot]