roleoroleo/yi-hack-Allwinner-v2 #992: Onvif AgentDVR

Watanakas, 2025-01-19:
Problem with Onvif and AgentDVR. In the previous version 0.3.3 the detection of new Onvif cameras works correctly. In the current version it does not detect it and gives an error 

Error in deserializing body of reply message for operation 'GetProfiles'.

Any suggestions, thanks.

roleoroleo, 2025-01-19:
I tested the cam now with AgentDVR and 0.3.3 and it works.
Please share your /tmp/onvif,conf (hide password if present).
When does the error appear?

Watanakas, 2025-01-20:
agent dvr does not recognize onvif camera, in new camera option and discover URL, receives 

Error in deserializing body of reply message for operation 'GetProfiles'.

is this the file you request?

onvif_simple_server.conf

model=Yi Hack
manufacturer=Yi
firmware_ver=0.3.3
hardware_id=BFUS
serial_num=BFUSY44xxxxxxxxxxxxx
ifs=wlan0
port=8181
scope=onvif://www.onvif.org/Profile/Streaming
scope=onvif://www.onvif.org/Profile/T
scope=onvif://www.onvif.org/hardware
scope=onvif://www.onvif.org/name
adv_enable_media2=0
adv_fault_if_unknown=0
adv_fault_if_set=0
adv_synology_nvr=0

user=root
password=Password

#Profile 1
name=Profile_1
width=640
height=360
url=rtsp://root:Password@%s/ch0_1.h264
snapurl=http://root:Password@%s:8181/cgi-bin/snapshot.sh?res=low&watermark=no
type=H264
audio_encoder=none
audio_decoder=NONE

#EVENT
events=3
#Event 0
topic=tns1:VideoSource/MotionAlarm
source_name=VideoSourceConfigurationToken
source_value=VideoSourceToken
input_file=/tmp/onvif_notify_server/motion_alarm
#Event 1
topic=tns1:RuleEngine/MyRuleDetector/PeopleDetect
source_name=VideoSourceConfigurationToken
source_value=VideoSourceToken
input_file=/tmp/onvif_notify_server/human_detection
#Event 2
topic=tns1:RuleEngine/MyRuleDetector/VehicleDetect
source_name=VideoSourceConfigurationToken
source_value=VideoSourceToken
input_file=/tmp/onvif_notify_server/vehicle_detection
#Event 3
topic=tns1:RuleEngine/MyRuleDetector/DogCatDetect
source_name=VideoSourceConfigurationToken
source_value=VideoSourceToken
input_file=/tmp/onvif_notify_server/animal_detection
#Event 4
topic=tns1:RuleEngine/MyRuleDetector/BabyCryingDetect
source_name=AudioSourceConfigurationToken
source_value=AudioSourceToken
input_file=/tmp/onvif_notify_server/baby_crying
#Event 5
topic=tns1:AudioAnalytics/Audio/DetectedSound
source_name=AudioSourceConfigurationToken
source_value=AudioSourceToken
input_file=/tmp/onvif_notify_server/sound_detection

roleoroleo, 2025-01-21:
I don't remember exactly the difference between your version and the current version (I mean onvif_simple_server).
Do you want to try a beta version with the new release of onvif_simple_server?
Maybe the problem is fixed.
In case. please give me your model suffix.

Watanakas, 2025-01-21:
The suffix 9.0.19 is y21ga, I think the problem may be the AgentDVR, since with Homeassistan's ONVIF it recognizes it and configures it correctly, I will downgrade to version 0.3.1 and wait for the next stable version.

roleoroleo, 2025-01-22:
Unfortunately onvif is not a standard,
Each manufacturer implements it as they like and there are no clear guidelines for implementation.
So it is absolutely normal to find incompatibilities between devices.
Here the beta version if you want to try it: [y21ga_0.3.3.tgz](https://github.com/user-attachments/files/18509909/y21ga_0.3.3.tgz)

closed 2025-05-23 by github-actions[bot]