roleoroleo/yi-hack-Allwinner-v2 #993: Feature Request: Read Only Mode

dreimer1986, 2025-01-28:
Hi, 
I wonder about something for a while now... Why don't you allow to set the Linux modifications to run on a secondary partition which is read only? A few days ago something strange happened and it's not the first time that similar conditions happened. We had a small power fluctuation and two of my three cams were dead because it took the filesystem together with the power loss. I fixed the filesystem, reinstalled the mod, the cam came back, but the rtsp server did not. All was fine on first look except the cam streams were unavailable. The SD filesystem was fine, but still I was completely unable to get the streams online again. (Yes I sent back it's config backup, still no dice.)

How about using a swapfs partition for the swap file and a read only partition for the Linux modifications? Of course we still need a r/w one for the screenshot archiving and some other things, but that filesystem being broken would not cause the cams to die on power loss so easily.

What do you think?

roleoroleo, 2025-01-28:
I've thought about it several times but I'm afraid it will become too complicated for not expert users.
The sd is now formatted with fat32 and everyone is more or less able to do it. And you can format it from the app.
With a complex partitioning the problems increase.

dreimer1986, 2025-01-28:
Yeah, the only way this could be sort of feasable for beginners is if the install procedure can format it for you. Anything else is close to impossible for ppl who never heard of Linux or WSL...

closed 2025-05-29 by github-actions[bot]