roleoroleo/yi-hack-Allwinner-v2 #995: Yi Home 1080p ai+ (y21ga) no sd reading and wifi psw error

a78p, 2025-02-07:
Hi to all the team, in someway i blocked, i don't know if bricked, my camera, it can't read sd card so no yi hack and wrong psw message in stock firmware. Some suggestion please? How to proceed?

roleoroleo, 2025-02-16:
If the problem is the sd card, replace it with a new one.
If the cam doesn't start, use the unbrick procedure.

roleoroleo, 2025-02-22:
> I can have a backup from another same camera I've, could be ok?

Yes, if it exactly the same model.

Use the soft way, it must work.
I don't have a video, what procedure is not clear? Hard or soft?

a78p, 2025-02-23:
Sorry, not clear both for me, but I'm looking at first at soft unbrick. For the soft unbrick I've many doubts and also some blocking points. I hope to be not so boring because I've many basic questions:

- My cameras Yi Home 1080p ai+ (y21ga) don't work with the last hack versions (i.e 0.3.3), but they work fine with 0.2.0 that was running for more than two years. Tried also some previous 0.3 version wkthout success. I don't understand why....
- To make the backup as in this [page](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)) , have I to decompress the [backup.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/11959592/backup.tar.gz) in a formatted SD or in the SD I have to copy previously also the Hack?
- The soft unbrick is possible only with Linux distro like Debian with "mtd-utils" and "u-boot-tools" installed? No way with Windows?
- In the page of the [soft unbrick description](https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick) where is written _" clone this repo in a Linux machine "_ do you mean, in the y21ga case for example, to download [these files](https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick/y21ga) in a Linux machine?
- Have I to delete something when you write "Remove init.sh from unbrick procedure" [here](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/26c00047edbd03d548b19dd95b7896d2c805e84c)
- where is written _" Copy your partition (mtdblockX.bin) in the folder corresponding to your model. mtdblock2.bin for rootfs partition, mtdblock3.bin for home partition and mtdblock4.bin for backup partition "_ where can I find mtdblock2.bin, mtdblock3.bin and mtdblock4.bin? in the backup from my camera? what's the _" corresponding folder of my camera "_ ? do you mean [this folder](https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick/y21ga) already dowloaded before?
- where can I find the unbrick folder to go in to run build command? 
- Have I to run the build command in a linux terminal?
- I would prefer to create at first an original partition " ./build.sh factory ", can I install later the hack with the standard installation copying the hack files in the SD and power on the camera?
- again what's the _" corresponding folder "_ of my camera where should I find "backup_XXX.gz (rootfs_XXX.gz and/or home_XXX.gz)" 
- check for errors it means to compare the size of the rootfs_XXX.gz file with mtdblock2.bin ? and home_XXX.gz with mtdblock3.bin ?
- at the end unzip the file in the sd card do you mean unzip this file rootfs_XXX.gz ?

roleoroleo, 2025-02-24:
> Sorry, not clear both for me, but I'm looking at first at soft unbrick. For the soft unbrick I've many doubts and also some blocking points. I hope to be not so boring because I've many basic questions:

Don't worry.

>     * My cameras Yi Home 1080p ai+ (y21ga) don't work with the last hack versions (i.e 0.3.3), but they work fine with 0.2.0 that was running for more than two years. Tried also some previous 0.3 version wkthout success. I don't understand why....

y21ga is my "develop" cam, I can assure you that it works properly with the latest release.

> * To make the backup as in this [page](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)) , have I to decompress the [...]

Yes, buf first you should check in your sd card because the hack procedure creates a backup when started. If you don't find any backup please let me know, I will check in my archive.

>     * The soft unbrick is possible only with Linux distro like Debian with "mtd-utils" and "u-boot-tools" installed? No way with Windows?

No way with Windows if you use the "hacked" option. No problem for "factory" option.

> * In the page of the [soft unbrick description](https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick) where is written _" clone this repo in a Linux machine "_ do you mean, in the [...]

No, it means that you have to open a shell in a linux machine and clone this git repository with the command
`git clone https://github.com/roleoroleo/yi-hack-Allwinner-v2.git`
This command will dowload the content of the repository, included the unbrick folder.

>     * Have I to delete something when you write "Remove init.sh from unbrick procedure" [here](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/26c00047edbd03d548b19dd95b7896d2c805e84c)

No.

> * where is written _" Copy your partition (mtdblockX.bin) in the folder corresponding to your model. mtdblock2.bin for rootfs partition, mtdblock3.bin for home partition and mtdblock4.bin for backup [...]

mtdblockN.bin are in your backup.
When the backup procedure runs it creates 8 files: from mtdblock0.bin to mtdblock7.bin

>     * where can I find the unbrick folder to go in to run build command?

When you clone the repo, a folder unbrick is created inside the yi-hack-Allwinner-v2 folder.

>     * Have I to run the build command in a linux terminal?

Yes, or a cmd prompt in Windows if you are creating a "factory" image.

>     * I would prefer to create at first an original partition " ./build.sh factory ", can I install later the hack with the standard installation copying the hack files in the SD and power on the camera?

Yes.

>     * again what's the _" corresponding folder "_ of my camera where should I find "backup_XXX.gz (rootfs_XXX.gz and/or home_XXX.gz)"

Inside the unbrick folder there are a lot of folder, one for each cam model.
Find the file corresponding to the partition you are restoring.
You should start to restore backup partition (mtdblock4). Usually it's enough.
So the file you have to find is backup_y21.gz

>     * check for errors it means to compare the size of the rootfs_XXX.gz file with mtdblock2.bin ? and home_XXX.gz with mtdblock3.bin ?

When you run the command, it could terminate with an error, Check the output text of the command.
And check if the final file created is zero lentgh.

>     * at the end unzip the file in the sd card do you mean unzip this file rootfs_XXX.gz ?

In your case, decompres backup_y21.ga in the sd card.

a78p, 2025-03-08:
Thank you very much for your detailed explanation, it looks more and more clear also for me now and for sure can help also other people not so expert as me.
Just to let you know, finally I didn't test this soft unbrick, so I don't know if is successful, because after I've requested support to Yi and they didn't answer, Amaz offered to me a refund.
Anyway thank you very much for your support and wish the best for your developments.

closed 2025-07-07 by github-actions[bot]