roleoroleo/yi-hack-Allwinner-v2 #996: [y211ga] Yi Home Camera 1080p: on weak wifi connection: sometimes no reconnect

luckyfreddy, 2025-02-10:
[y211ga] Yi Home Camera 1080p: on weak wifi connection: sometimes no reconnect

Hello together.

Many thanks for the great work with this firmware. At one camera unfortunately the wifi signal is weak and cannot be optimized (its in an birds nest box). Sometimes (my guess when the neighbors using her wifi) our wifi-signal seemed to be to weak and the connection break. Unfortunately the cam sometimes does not connect to wifi again and has to be rebooted. 

In another thread I've found the following script, wich I edited for myself and it seemed to work for me and reconnect the cam again:

`while true; do if ! wpa_cli -i wlan0 status | grep -q "wpa_state=COMPLETED"; then echo -e "$(date): CONNECTION LOST\n\n$(wpa_cli -i wlan0 status)\n\n$(wpa_cli interface)\n\n$(ps)\n-----\n\n" >> /tmp/sd/yi-hack/wlanlog.txt; sleep 2; wpa_cli -i wlan0 reconfigure; fi; sleep 10; done &`

I have added this code at the end of the file "/tmp/sd/yi-hack/script/mqtt_advertise/startup.sh" and rebooted the cam. To be safe I also restart the cam every six hours with active cronjob.

Because the cam is buildin at an birds nest box the physical access to it is a little bit complicated. Thats why I tried to reproduced the behaviour at another cam with mobile tethering (placed far away from the cam) and additional with deauth attacks to my mobile hot spot to simulate. Unfortunately I cannot reproduce the lack of reconnection. But even at the real cam this behaviour happens only sometimes a day (sometimes multiple times a day, but sometimes after several days of working). 
Now the connection seemed to be more stable (or rather faster reconnecting) and the cam does not need to be an manual reboot again.

Here are the /proc/net/wireless from the real cam, on wich it happens:
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   38.  -72.  -256        0      0      0      2  32468        0
  p2p0: 0000    0     0     0        0      0      0      0      0        0
WiFi Strength (according to web ui): 58 %

With the workaround showing above there was following interventions, where the script reconnected the cam successfully:
Sat Feb  8 08:47:35 CET 2025: CONNECTION LOST
Sat Feb  8 08:47:47 CET 2025: CONNECTION LOST
Sat Feb  8 08:48:51 CET 2025: CONNECTION LOST
Sat Feb  8 08:49:04 CET 2025: CONNECTION LOST
Sat Feb  8 08:50:07 CET 2025: CONNECTION LOST
Sat Feb  8 08:50:19 CET 2025: CONNECTION LOST
Sat Feb  8 08:50:32 CET 2025: CONNECTION LOST
Sat Feb  8 08:50:44 CET 2025: CONNECTION LOST
Sat Feb  8 08:50:57 CET 2025: CONNECTION LOST
Sat Feb  8 08:51:10 CET 2025: CONNECTION LOST
Sat Feb  8 08:51:33 CET 2025: CONNECTION LOST
Fri Apr 17 22:17:30 CEST 2020: CONNECTION LOST
Sun Feb  9 09:30:26 CET 2025: CONNECTION LOST
Sun Feb  9 09:33:09 CET 2025: CONNECTION LOST
Sun Feb  9 18:10:06 CET 2025: CONNECTION LOST
Sun Feb  9 18:11:59 CET 2025: CONNECTION LOST

I faced this behaviour with FW 0.2.6, later with FW 0.3.0 and now with FW 0.3.3. For a while I replaced the cam with another one (cause i thought that they were broken) and I changed the sd card, as well. In the end it tourned out, that only the wifi connection were the problem - the cam itself recorded on local storage (even without reconnecting).

BELOW THIS LINE the output is only from my test constellation:

And here at the test-cam where I simualted the behaviour:
root@yicamtestlocalnetwork:~# cat /proc/net/wireless
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   34.  -76.  -256        0      0      0      0      6        0
  p2p0: 0000    0     0     0        0      0      0      0      0        0
WiFi Strength (according to web ui): 48 %

root@yicamtestlocalnetwork:~# grep LOST /tmp/sd/yi-hack/wlanlog.txt
Mon Feb 10 10:01:22 CET 2025: CONNECTION LOST
Mon Feb 10 10:01:35 CET 2025: CONNECTION LOST
Mon Feb 10 10:01:48 CET 2025: CONNECTION LOST
Mon Feb 10 10:13:47 CET 2025: CONNECTION LOST
Mon Feb 10 10:14:00 CET 2025: CONNECTION LOST
Mon Feb 10 10:14:12 CET 2025: CONNECTION LOST
Mon Feb 10 10:14:25 CET 2025: CONNECTION LOST
Mon Feb 10 10:14:38 CET 2025: CONNECTION LOST
Mon Feb 10 10:14:50 CET 2025: CONNECTION LOST
Mon Feb 10 10:16:03 CET 2025: CONNECTION LOST
Mon Feb 10 10:16:36 CET 2025: CONNECTION LOST
Mon Feb 10 10:17:08 CET 2025: CONNECTION LOST
Mon Feb 10 10:17:41 CET 2025: CONNECTION LOST
Mon Feb 10 10:18:14 CET 2025: CONNECTION LOST

In detail:
```
root@yicamtestlocalnetwork:~# cat /tmp/sd/yi-hack/wlanlog.txt
Mon Feb 10 10:01:22 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:01:35 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:01:48 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:13:47 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:14:00 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:14:12 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:14:25 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:14:38 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:14:50 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:16:03 CET 2025: CONNECTION LOST

wpa_state=SCANNING
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:16:36 CET 2025: CONNECTION LOST

wpa_state=ASSOCIATING
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
-----

Mon Feb 10 10:17:08 CET 2025: CONNECTION LOST

[cutted the process table]
-----

Mon Feb 10 10:17:41 CET 2025: CONNECTION LOST

[cutted the process table]

-----

Mon Feb 10 10:18:14 CET 2025: CONNECTION LOST

wpa_state=SCANNING
address=b4:fb:e3:dc:1c:da
uuid=ef0829e2-a59d-52b7-af04-72f1c966d6db

Selected interface 'wlan0'
Available interfaces:
wlan0

[cutted the process table]
```

Even the workaround seemed to work for me, I'm not sure, if it is the best way and even the "/tmp/sd/yi-hack/script/mqtt_advertise/startup.sh" could be possible recreated update or maybe web ui on configuration so that the workaround possible don't stay.

Is there anything to check or configure better? The problem occurs with swap enabled and swap disabled; there seemed always to be ~12MB of free memory.

Thanks for help and greetings LF.

liebefuroscar, 2025-02-12:
I'm having the same problem on my "y623". Wifi signal at over 80% and it keeps disconnecting. When it does it's not able to reconnect. I've implemented your script and it seems to be fine.
I'll update in a few days with feedback. Thanks

roleoroleo, 2025-02-16:
The script is a good idea.
The best position to run it is the file
/tmp/sd/yi-hack/startup.sh
It's launched at the end of the init procedure, the last line of system.sh and is not overwritten during the upgrade.

liebefuroscar, 2025-02-17:
Update 17/2.
I've been using the script for almost a week now. I haven't had any disconnections lasting more than 3 minutes. (I made an automation to reboot the camera if there was a disconnection to Frigate of more than 3 minutes and it has not been executed once). My question is what will happen when we update the app to a new firmware.Could it be included in the firmware natively?

luckyfreddy, 2025-02-17:
Since today it looks like that the problem is solved by this workaround. In the log I can see, that in some cases the script has reconnected the cam to wifi successfully. Unfortunately today the connection get lost again and this time the workaround hasn't repaired the connection. Only the cronjob, which reboots the cam every 6 hours, has solved the issue this time.
Maybe an `wpa_cli -i wlan0 reconfigure` is not enough in some cases. 

In the written logfile /tmp/sd/yi-hack/wlanlog.txt I could see:
```
Mon Feb 17 10:01:32 CET 2025: CONNECTION LOST
[...]
Mon Feb 17 11:59:53 CET 2025: CONNECTION LOST
```
for 549 times. 

Even the following entry is continusly logged except from date and time:
`Mon Feb 17 10:01:32 CET 2025: CONNECTION LOST

wpa_state=DISCONNECTED
address=9c:1c:37:87:2e:03
uuid=8f9ca0a8-71e5-50eb-aa62-2be323e623ac

Selected interface 'wlan0'
Available interfaces:
wlan0

  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  450 root         0 SW   [irq/302-sunxi-m]
  452 root         0 SW   [irq/166-sdc0 cd]
  453 root         0 SW   [irq/303-sunxi-m]
  525 root         0 SW<  [kworker/0:1H]
  527 root         0 SW<  [bioset]
  530 root         0 SW   [mmcqd/0]
  546 root         0 SWN  [jffs2_gcd_mtd4]
  599 root         0 SW<  [phy0-atbm_wq]
  600 root         0 SW   [phy0-usb_atbm_b]
  653 root      2380 S    ./dispatch
 1439 root     40176 S    ./rmm
 1710 root      3056 S    ./mp4record
 1761 root       968 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1777 root       892 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sb
 1867 root       880 S    mqttv4
 1917 root       968 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/u
 2008 root      5464 S<   rRTSPServer -m y211ga -r both -a no -p 554
 2015 root       992 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.
 2052 root       968 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 2359 root       992 S    {startup.sh} /bin/sh /tmp/sd/yi-hack/script/mqtt_adv
16731 root         0 SW   [kworker/u2:0]
24879 root         0 SW   [kworker/0:0]
25869 root         0 SW   [kworker/u2:2]
31239 root         0 SW   [kworker/u2:1]
31499 root       980 S    /bin/sh -c sleep 40; /tmp/sd/yi-hack/script/ftppush.
31501 root       980 S    sleep 40
31648 root         0 SW   [kworker/0:2]
31662 root         0 SW   [kworker/0:3]
31663 root         0 SW   [kworker/0:4]
31679 root      1448 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
31680 root      1048 S    sh -c /home/app/script/wifidhcp.sh
31681 root      1048 S    {exe} ash /home/app/script/wifidhcp.sh
31699 root      1048 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -
31735 root       980 S    sleep 10
31737 root         0 Z    [init]
31743 root       980 R    ps`

Unfortunately there is no "SCANNING" or "CONNECTING" to see. Maybe I should try to change `wpa_cli -i wlan0 reconfigure` into `/sbin/reboot` or add this as an option after three successful connection attempts. I think I will try this soon at my test cam and in the meantime I'll set the reboot-cronjob-interval back to three hours to be safer.

luckyfreddy, 2025-02-18:
**Update:** Since I want to avoid restarting the camera (cause recording on sd card continues even without wifi), I have expanded and test the the workaround at my test camera a bit. Beforce sending an `wpa_cli -i wlan0 reconfigure` the interface would be set down and after a second set up again. I hope that this will help to provide the last failures. Now I have installed this workaround at the nestbox camera with the connection problems. I will continue to observe. To be safe and not get locked out, I have a cron job running on the camera, which restarts the camera every three hours again.

My newer extended workaround:
`while true; do if ! wpa_cli -i wlan0 status | grep -q "wpa_state=COMPLETED"; then echo -e "$(date): CONNECTION LOST\n\n$(wpa_cli -i wlan0 status)\n\n$(wpa_cli interface)\n\n$(ps)\n-----\n\n" >> /tmp/sd/yi-hack/wlanlog.txt; sleep 2; /sbin/ifconfig wlan0 down; sleep 1; /sbin/ifconfig wlan0 up; sleep 1; wpa_cli -i wlan0 reconfigure; fi; sleep 10; done &`

As from @roleoroleo recommended I added the line above in a new file `/tmp/sd/yi-hack/startup.sh` - together with an `#!/bin/sh` (shebang) in first line, cause the file didn't exist before. And (of course) I removed my first entry in `/tmp/sd/yi-hack/script/mqtt_advertise/startup.sh` (which I mentioned in the first post).

I'll observe and report.

liebefuroscar, 2025-02-18:
it's great, but I see some "issues":  

- if the connection is not restored, the script enters an infinite loop without a clear solution. this can cause the device to keep trying to reconnect unsuccessfully, affecting performance.  
- the log file (`/tmp/sd/yi-hack/wlanlog.txt`) grows uncontrollably.  
- although the `"wpa_state=COMPLETED"` status indicates that the connection is established, it does not guarantee actual internet connectivity. it might be useful to include an additional check with a ping to an external server.  

i'm considering the following improvements:  

- limit the number of reconnection attempts and, if a maximum is reached, restart the device or generate an alert.  
- implement log rotation to prevent the file from becoming too large.  
- add a real connectivity check using a ping.

luckyfreddy, 2025-02-19:
Now I have put the workaround into a script and optimized it further.

Improvements:
- the script tries up to six times to reestablished wifi connection and (if this fails) restart the camera
- it never keeps more than 150 characters in the logfile (to prevent full storage)
- the output at the logfile is reduced to wifi state messages and messages from script
- the path and naming of the logfile is now similar to existing "Debug log” from the UI

I saved the following code as file /tmp/sd/yi-hack/wifi_failsafe_watchdog.sh and set this file at the /tmp/sd/yi-hack/startup.sh, so that the /tmp/sd/yi-hack/startup.sh now looks like this (of course after removing the previous workaround at this file):

**/tmp/sd/yi-hack/startup.sh**
```
#!/bin/sh
/tmp/sd/yi-hack/wifi_failsafe_watchdog.sh
```

**/tmp/sd/yi-hack/wifi_failsafe_watchdog.sh**
```
#!/bin/sh

LOGFILE="/tmp/sd/hack_wififailsafe.log"

while true; do
    if ! wpa_cli -i wlan0 status 2>&1 | grep -q "wpa_state=COMPLETED"; then
        if [ -e "$LOGFILE" ]; then
                /tmp/sd/yi-hack/usr/bin/tail -n 145 "$LOGFILE" > "$LOGFILE.tmp" && mv "$LOGFILE.tmp" "$LOGFILE"
            fi
		echo -e "$(date): Wifi connection lost:\n$(wpa_cli -i wlan0 status 2>&1)" >> "$LOGFILE"
        failsafecounter=$((failsafecounter + 1))
        if [ "$failsafecounter" -ge 6 ]; then
            echo -e "$(date): Wifi connection still could't be restored. Restarting." >> "$LOGFILE"
			reboot
        else
            echo -e "$(date): Attempting reconnect." >> "$LOGFILE"
			sleep 2
            ifconfig wlan0 down
            sleep 1
            ifconfig wlan0 up
            sleep 1
            wpa_cli -i wlan0 reconfigure
        fi
    fi
    sleep 10
done &
```
I decided not to check the internet access in favor of the established wifi connection, because this workaround was originally intended to keep the wifi connection "stable" and secure. Without wifi connection it is neither possible to see if the camera is still alive nor access to the web interface/ssh. With established wifi connection access to the web interface/ssh should be possible for further measures possible. In addition, an internet disruption would then also have an impact on the camera itself if the script would restart the camera, for example. So I think foucussing on an established wifi connection should be the best in that case.

roleoroleo, 2025-02-24:
I recommend using /tmp/sd/yi-hack/bin for your scripts or binaries
and /tmp/sd/yi-hack/startup.sh to launch it.
In case of upgrade, these files won't be deleted or modified.

Another little change: add a & at the end of the line:
```
#!/bin/sh
/tmp/sd/yi-hack/wifi_failsafe_watchdog.sh &
```
to allow startup.sh to terminate.

luckyfreddy, 2025-02-24:
> I recommend using /tmp/sd/yi-hack/bin for your scripts or binaries

Good idea. I will move the files to `/tmp/sd/yi-hack/bin`. I have not done this, because I was not sure if the folders getting delete while upgrades. What will happen to the `/tmp/sd/yi-hack/startup.sh` after upgrading? Will the script-command be persistent?

> Another little change: add a & at the end of the line:

The `/tmp/sd/yi-hack/startup.sh` script would be terminated even without the `&` at the end of `/tmp/sd/yi-hack/wifi_failsafe_watchdog.sh &`, because the `while` command at the script already contains the `&` (after `done`) which send the complete while statement to background.

Intermediate status of my script-test: so far there have been no connection problems. I will continue to monitor it.

roleoroleo, 2025-02-28:
> What will happen to the /tmp/sd/yi-hack/startup.sh after upgrading? Will the script-command be persistent?

Yes.

luckyfreddy, 2025-03-05:
> **/tmp/sd/yi-hack/startup.sh**
> 
> ```
> #!/bin/sh
> /tmp/sd/yi-hack/bin/wifi_failsafe_watchdog.sh
> ```
> 
> **/tmp/sd/yi-hack/bin/wifi_failsafe_watchdog.sh**
> 
> ```
> #!/bin/sh
> 
> LOGFILE="/tmp/sd/hack_wififailsafe.log"
> 
> while true; do
>     if ! wpa_cli -i wlan0 status 2>&1 | grep -q "wpa_state=COMPLETED"; then
>         if [ -e "$LOGFILE" ]; then
>                 /tmp/sd/yi-hack/usr/bin/tail -n 145 "$LOGFILE" > "$LOGFILE.tmp" && mv "$LOGFILE.tmp" "$LOGFILE"
>             fi
> 		echo -e "$(date): Wifi connection lost:\n$(wpa_cli -i wlan0 status 2>&1)" >> "$LOGFILE"
>         failsafecounter=$((failsafecounter + 1))
>         if [ "$failsafecounter" -ge 6 ]; then
>             echo -e "$(date): Wifi connection still could't be restored. Restarting." >> "$LOGFILE"
> 			reboot
>         else
>             echo -e "$(date): Attempting reconnect." >> "$LOGFILE"
> 			sleep 2
>             ifconfig wlan0 down
>             sleep 1
>             ifconfig wlan0 up
>             sleep 1
>             wpa_cli -i wlan0 reconfigure
>         fi
>     fi
>     sleep 10
> done &
> ```

Short update: after installing the last workaround from February 19, 2025 (https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/996#issuecomment-2668576747) I have not faced any wifi connection problems again. Just in one case the connetion got lost and the workaround has successfully reconnected the camera again in short time. In my case the workaround seemed to help well. It runs well on four cameras (all y211ga).

roleoroleo, 2025-03-13:
I could add this script inside wd_rtsp.sh

roleoroleo, 2025-03-14:
Thank you for your code:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/60b2d980e240679e87671eab37edfaab6a31d1e1

closed 2025-10-19 by github-actions[bot]