roleoroleo/yi-hack-Allwinner-v2 #997: [y211ga] Yi Home Camera 1080p: Any way to force "color mode" after night-vision on medium light?

luckyfreddy, 2025-02-13:
I have observed that in many cases after using night vision the cam doesn't set back to color vision - even when disabling IR/night vision mode and the brightness is actually sufficient. When restarting the cam the color vision appears correctly and it is not to dark. I think, that some tresholds won't be reached or exceeded so that the camera remains in black/white mode. Is there any possibility to activate manually the color-mode without rebooting the cam? I activate/deactivated IR/night vision via curl call http://IP-CAM:8080/cgi-bin/camera_settings.sh?ir=yes which works great. But even with deactivated IR/night vision the black/white-view stays sometimes active for a while.

Thanks for any ideas.

roleoroleo, 2025-02-16:
Sorry but there is no way to configure ir and cut filter.
They are handled by the Yi processes.

luckyfreddy, 2025-02-17:
Even if there is no way to configure the filter - is there maybe a way to send an command to manually trigger/force the color mode? Because the ir itself I could control with the 
`wget -q -O - 'http://IP-CAM:8080/cgi-bin/camera_settings.sh?ir=yes'` and ` wget -q -O - 'http://IP-CAM:8080/cgi-bin/camera_settings.sh?ir=no'` at runtime and this call is made from homematic (home automation) when sensing low or high light at an dedicated light-sensor. An manual trigger (even an ssh-command) would be enough for me, even if the camera can't do it by itself.

luckyfreddy, 2025-02-18:
At this time I reboot the camera and then the firmware set the color mode (and shut off cut filter). Is it maybe possible to restart only the yi-service without rebooting to speed it up? Or is there any kind of sniffer which i can run (when camera is in low light) to sniff commands from the firmware itself, which maybe could help in that case?

roleoroleo, 2025-02-22:
With camera_settings command you can enable/disable IR where enable means that Yi handles when IR is switched on or off.

The process that handles IR is rmm and cannot be restarted.
If you kill it and restart it, it's unable to inititialize again the cam again.
Reboot is required.
You can try, maybe your cam is different.

> Or is there any kind of sniffer which i can run (when camera is in low light) to sniff commands from the firmware itself, which maybe could help in that case?

I didn't check if there is an IPC command between processes when IR switches.
You can try 2 different way:

- Remove temporarily rmm from automatic start and start it manually from shell. It shows a very verbose output.
- ipc_sniff can sniff the IPC queue /ipc_dispatch (it's the main queue used but not the only one)

```
root@yi-hack-h52ga:~# ls /dev/mqueue/
ipc_cloud
ipc_dispatch
ipc_dispatch_worker
ipc_mdns
ipc_p2p
ipc_rcd        
ipc_rmm
ipc_rtmp
```

[ipc_sniff.gz](https://github.com/user-attachments/files/18923046/ipc_sniff.gz)

luckyfreddy, 2025-02-24:
With your first method (temporarily removing rmm from automatic start and start it manually from shell) get the following codes, which should be interesting:

**Camera switched to black/white mode:**
```
rmm.c(auto_loop_func-2721)[07:57:05.554]:do isp switch, previous gIsGraySwitch is 0, now is 1
[ISP]ISP select ir config
[ISP]find sp2305_mipi_1936_1096_20_0 [sp2305_mipi_init_night] isp config
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 1
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 1
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 1
```

**Camera switched to color mode:**
```
rmm.c(auto_loop_func-2721)[07:57:17.368]:do isp switch, previous gIsGraySwitch is 1, now is 0
[ISP]find sp2305_mipi_1936_1096_20_0 [sp2305_mipi_init_day] isp config
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 0
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 0
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 0
```

**Camera switched to black/white mode again:**
```
rmm.c(auto_loop_func-2721)[07:57:45.809]:do isp switch, previous gIsGraySwitch is 0, now is 1
[ISP]ISP select ir config
[ISP]find sp2305_mipi_1936_1096_20_0 [sp2305_mipi_init_night] isp config
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 1
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 1
debug  : cedarc <H264SetParameterVer2:7548>: VENC_IndexParamIsNightCaseFlag: is_night_case = 1
```

With the second method I faced some difficulties using ipc_sniff. When I was starting, for example `/tmp/sd/yi-hack/ipc_sniff /dev/mqueue/ipc_dispatch` then I get an `Can't open mqueue /dev/mqueue/ipc_dispatch. Error: Permission denied`. Same with the other queues. I even tried to set the command `/tmp/sd/yi-hack/ipc_sniff /dev/mqueue/ipc_dispatch >> /tmp/sd/yi-hack/ipc_dispatch.log` into the first line of `/tmp/sd/yi-hack/script/system.sh`. But unfortonately even then there would not be logged. With `ps` I tried to find an process called `ipc_multiplexer` or similar. But I could not find someone. What I am doing wrong?

roleoroleo, 2025-02-24:
Based on this log the process that handles ir is rmm.
We could try to switch on/off the gpio.

About ipc_sniff, this should work:
`/tmp/sd/bin/ipc_sniff /ipc_dispatch`

For ipc_dispatch queue you can use a multiplexed copy, if you don't want to trouble other processes:
`/tmp/sd/bin/ipc_sniff /ipc_dispatch_5`

luckyfreddy, 2025-03-05:
> Based on this log the process that handles ir is rmm. We could try to switch on/off the gpio.
Do you mean via `printf 0>/sys/class/gpio/XXX` ? Which gpio-path could be the right one?

> About ipc_sniff, this should work: `/tmp/sd/bin/ipc_sniff /ipc_dispatch`
It worked from command invocation perspective. Unfortunately I could not get an usable result.

My observations at /ipc_dispatch:

**Normal messages (background noise)**
```
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 04 00 00 00 22 40 01 00 00 00 00 00
```

**Probable changes to color mode**
```
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 02 00 00 00 7c 00 7c 00 00 00 00 00
```
**another time when switching to color mode**
```
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00
```

**First I thought the following would be an message captured while switching to black/white mode. But the message appears even without changes and sometimes with no apparent cause.**
```
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 10 00 00 00 e7 00 e7 00 00 00 00 00
```
For time reasons I am not able to dive in deeper at the moment. After observing I am not sure if the monitored codes stays really in connection to the color-mode and black/white-mode. I could not get causal readings when switching. So if there is no possible way to set the modes with an gpio-switch than this case can be closed.

roleoroleo, 2025-03-06:
2nd message is motion detected start
3rd message is motion detected end
About 1st and 4th, they are very common but I wasn't able to connect to a specfic event.
I think there is no need to send this message to a queue because rmm is source and destination.

About gpios, these numbers are related to IR:
```
21 IR filter night
22 IR filter day
```
Nothing about IR on and off.
But I don't know if they are specific for a model.
I tested them on my y21ga.
I can share the binary to send gpios if you want to try; you can see the code in src/speaker
[gpio.gz](https://github.com/user-attachments/files/19110936/gpio.gz)
but pay attention: move a gpio can damage the cam.

luckyfreddy, 2025-03-10:
I will test it in a few days, cause I am a little bit busy right now and have no access to the test camera. I will report the result.

luckyfreddy, 2025-03-14:
> About gpios, these numbers are related to IR:
> 
> ```
> 21 IR filter night
> 22 IR filter day
> ```
> 
> Nothing about IR on and off. But I don't know if they are specific for a model. I tested them on my y21ga.

I have tested it now. Setting the gpio works even for the y211ga. You can hear the "click" and see an difference at the image. Unfortunatelly the image stays in black/white even when the cut filter gpio is set to day. So I think it stays at the point you said in your first answer:
> They are handled by the Yi processes.

luckyfreddy, 2025-03-16:
I close this issue because it seems not to be possible to control this behaviour from outside official camera firmware.

closed 2025-03-16 by luckyfreddy