roleoroleo/yi-hack-Allwinner-v2 #1001: remove timestamp

openmotion, 2025-02-20:
hello 
i want to remove the timestamp in the video, is there an option to do this ?
thanks

roleoroleo, 2025-02-24:
No, sorry, is not possible.
I already tried in different ways but I have not succeeded.

roleoroleo, 2025-02-28:
Ok, but I can't use this procedure. Many people want to see the timestamp.
I should put in the hack another rmm binary...

roleoroleo, 2025-03-15:
> I managed to remove both the timestamp and bitmap in the video.
> But in order to do this you need to decompile the rmm binary with ghidra.
> Surprisingly it's not very hard because there's a lot of log so it's easy to find where the bitmap and timestamp are added to the video.
> You need to find which function do this and remove the call to her.

Using ghidra, did you find if the osd timestamp depends on some variable in the g_rmm_info struct?

closed 2025-07-19 by github-actions[bot]