roleoroleo/yi-hack-Allwinner-v2 #1007: [Bug] Enabling/disabling via Web Service automatically turns off motion tracking on Yi 2K Pro (Y623)

iMatux, 2025-03-12:
**Issue Description**

When I use a web service to enable or disable the camera (e.g., the set configuration of camera settings), the motion tracking automatically turns off.

Steps to Reproduce:
	1.	Send the following request via a browser or an HTTP client: http://<IP_ADDRESS>/cgi-bin/camera_settings.sh?switch_on=yes or with ?switch_on=no
	2. Authentication is correctly set up and working.
	3.	The camera turns on/off as expected, but motion tracking is automatically disabled.
	4.	This issue did not occur on my previous Yi 1080p camera but started when I switched to the Yi 2K Pro (Y623).
	5.	The issue does not happen when toggling the camera from the Kami app or the hack’s web interface.

If you need a dump, logs, or any other information, I’m happy to provide it. Let me know how I can help debug this.

roleoroleo, 2025-03-13:
What do you mean with "motion sensor is automatically disabled"?
Where? Inside the app? Web page?

iMatux, 2025-03-14:
You can see this both on the page=camera_settings page of the hack and in the Kami/Yi app. After sending a POST request to any web service to set a parameter (e.g., led=yes/no or any other parameters), motion detection is automatically disabled., if I refresh the camera_settings page, motion detection will no longer be active I can confirm this because I stop receiving notifications via MQTT and from the official app, meaning it turns off everywhere.

There is a glitch in the app that might be useful to analyze: after making a POST request to set a parameter, when I enter the “Detection Alert” section in the app, for about 1-2 seconds, it briefly shows as set to “Kami Cloud AI detections,” then switches to regular motion detection. I’m not sure if this is a default behavior, especially since in the AllWinner interface, I can see that all three AI detections, as well as motion detection, are turned off.

If motion detection is disabled, simply opening the app automatically re-enables the normal motion detection (without AI).
However, if I turn off the Yi camera via the camera_settings page or through the app, motion detection does not get disabled.

https://github.com/user-attachments/assets/86d78b50-b08f-4a08-aaa9-37918502ca65

closed 2025-03-15 by roleoroleo