roleoroleo/yi-hack-Allwinner-v2 #1010: Storm of Name Resolution for the MQTT Broker by Yi-Hack-Cameras on network

outofsight, 2025-03-26:
My LAN DNS Servers receive something like 30.000 DNS Name resolution every hour from each Yi-Hack-Camera, i.e. 150.000 request from all cameras.

Not sure if has already happened or started after some "recent" firmware upgrade.

It's something that is enought to rate-limit the DNS server and so preventing regular network operation.

Firmware Version | 0.3.3
Base Version | 9.0.19.02_202007141512
Model Suffix | y21ga

and also

Firmware Version | 0.3.9
Base Version | 8.1.0.0A_202001211401
Model Suffix | y30qa
Serial Number | BFUS

Thanks for any input.

outofsight, 2025-03-31:
The issue was related to the fact that the DNS name of broker was not resolving properly.

But this issue had this result only on yi-hacked cameras.
Event under ordinary conditions there are many resolution of that DNS name.

I suspect that the resolution is not cached in any way on yi-hack.

roleoroleo, 2025-04-05:
It could be mqttv4.
It checks the connection (and tries to reconnect) every 500 ms.

outofsight, 2025-04-05:
possible to cache the positive or negative DNS resolution according to TTL or at least for says 1min?

closed 2025-08-04 by github-actions[bot]