roleoroleo/yi-hack-Allwinner-v2 #1013: RTSP Connection refused in Yi Dome Guard YRS 12.1.47*

sergioruizsan, 2025-04-09:
Hello!

First of all thanks to @roleoroleo and the other contributors for your work mantaining this repo. This is super nice to have for local video recording :)

I've installed the yi-hack firmware to a Yi Dome Guard - YRS with firmware 12.1.47.05_202411081421. The Yi Home mobile app works as expected, it can stream video. But for some reason I can't get any snapshot or RTSP stream low or high resolution working locally (rtsp://192.168.1.10/ch0_1.h264). I've tried many configuration options, the following "Standard RTSP server" is the only one that gives me more output logs to debug. It seems the server is rejecting the connection, but don't know why. There is also no auth user/password set.

**System information:**
[image]

**VLC error:**
[image]

**FFmpeg log:**
```
ffmpeg -v debug -rtsp_transport tcp -i rtsp://192.168.1.10:554/ch0_1.h264 -c copy test.mp4
ffmpeg version 7.1.1 Copyright (c) 2000-2025 the FFmpeg developers
  built with Apple clang version 16.0.0 (clang-1600.0.26.6)
  configuration: --prefix=/opt/homebrew/Cellar/ffmpeg/7.1.1_1 --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags='-Wl,-ld_classic' --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libaribb24 --enable-libbluray --enable-libdav1d --enable-libharfbuzz --enable-libjxl --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox --enable-audiotoolbox --enable-neon
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
  libpostproc    58.  3.100 / 58.  3.100
Splitting the commandline.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'debug'.
Reading option '-rtsp_transport' ... matched as AVOption 'rtsp_transport' with argument 'tcp'.
Reading option '-i' ... matched as input url with argument 'rtsp://192.168.1.10:554/ch0_1.h264'.
Reading option '-c' ... matched as option 'c' (select encoder/decoder ('copy' to copy stream without reencoding)) with argument 'copy'.
Reading option 'test.mp4' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option v (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url rtsp://192.168.1.10:554/ch0_1.h264.
Successfully parsed a group of options.
Opening an input file: rtsp://192.168.1.10:554/ch0_1.h264.
[tcp @ 0x600003dd0000] No default whitelist set
[tcp @ 0x600003dd0000] Original list of addresses:
[tcp @ 0x600003dd0000] Address 192.168.1.10 port 554
[tcp @ 0x600003dd0000] Interleaved list of addresses:
[tcp @ 0x600003dd0000] Address 192.168.1.10 port 554
[tcp @ 0x600003dd0000] Starting connection attempt to 192.168.1.10 port 554
[tcp @ 0x600003dd0000] Connection attempt to 192.168.1.10 port 554 failed: Connection refused
[tcp @ 0x600003dd0000] Connection to tcp://192.168.1.10:554?timeout=0 failed: Connection refused
[in#0 @ 0x6000033d0a00] Error opening input: Connection refused
Error opening input file rtsp://192.168.1.10:554/ch0_1.h264.
Error opening input files: Connection refused
```

Anyone knows what could be happening? 
Thanks in advance!

roleoroleo, 2025-04-10:
Please, run the server in debug mode:
`rRTSPServer -m r35gb -r both -a aac -p 554 -d 15`

sergioruizsan, 2025-04-12:
Hi @roleoroleo, thanks for replying.

The following is all I got. I don't know if I need to trigger any other command on client side while debugging. Do I? I tried with with `ffmpeg -v debug -rtsp_transport tcp -i rtsp://192.168.1.10:554/ch0_1.h264 -c copy test.mp4` or including in the path "r35gb":  ffmpeg -v debug -rtsp_transport tcp -i rtsp://192.168.1.10:554/r35gb -c copy test.mp4. But didn't make any difference in the output.

```
root@yi-c212:~# killall wd.sh
killall: wd.sh: no process killed
root@yi-c212:~# killall rRTSPServer
root@yi-c212:~# rRTSPServer -m r35gb -r both -a aac -p 554 -d 15
1744474611265: the size of the buffer is 1048944
1744474611268: capture - mapping file fshare_frame_buf, size 1048944, to b6cb3000
1744474611269: capture - closing the file fshare_frame_buf
1744474611270: capture - frame header size = 26
1744474611271: capture - starting capture main loop
1744474611271: capture - buf_idx_end != input_buffer.buffer + input_buffer.offset + i
1744474611273: capture - buf_idx_end != input_buffer.buffer + input_buffer.offset + i 
...
1744474612503: capture - buf_idx_end != input_buffer.buffer + input_buffer.offset + i
1744474612504: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612504: capture - frame_sync == 0
1744474612514: capture - buf_idx_end == buf_idx_end_prev
1744474612525: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612525: capture - frame_sync == 0
1744474612535: capture - buf_idx_end == buf_idx_end_prev
1744474612545: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612545: capture - frame_sync == 0
1744474612555: capture - buf_idx_end == buf_idx_end_prev
1744474612566: capture - buf_idx_end == buf_idx_end_prev
1744474612576: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612576: capture - frame_sync == 0
1744474612586: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612586: capture - frame_sync == 0
1744474612596: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612596: capture - frame_sync == 0
1744474612606: capture - buf_idx_end == buf_idx_end_prev
1744474612617: capture - buf_idx_end == buf_idx_end_prev
1744474612627: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612627: capture - frame_sync == 0
1744474612637: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612637: capture - frame_sync == 0
1744474612647: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612647: capture - frame_sync == 0
1744474612657: capture - buf_idx_end == buf_idx_end_prev
1744474612668: capture - buf_idx_end == buf_idx_end_prev
1744474612678: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612678: capture - frame_sync == 0
1744474612688: capture - buf_idx_end == buf_idx_end_prev
1744474612698: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612698: capture - frame_sync == 0
1744474612708: capture - buf_idx_end == buf_idx_end_prev
1744474612718: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612719: capture - frame_sync == 0
1744474612729: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612729: capture - frame_sync == 0
1744474612739: capture - buf_idx_end == buf_idx_end_prev
1744474612749: capture - buf_idx_end == buf_idx_end_prev
1744474612759: capture - buf_idx_end == buf_idx_end_prev
1744474612769: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612770: capture - frame_sync == 0
1744474612780: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612780: capture - frame_sync == 0
1744474612790: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612790: capture - frame_sync == 0
1744474612800: capture - buf_idx_end == buf_idx_end_prev
1744474612810: capture - buf_idx_end == buf_idx_end_prev
1744474612820: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612820: capture - frame_sync == 0
1744474612831: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612831: capture - frame_sync == 0
1744474612845: capture - fhs[i].len > input_buffer.size - input_buffer.offset
[... 14 lines cut ...]
1744474612947: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612947: capture - frame_sync == 0
1744474612957: capture - buf_idx_end == buf_idx_end_prev
1744474612967: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612968: capture - frame_sync == 0
1744474612978: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474612978: capture - frame_sync == 0
1744474612988: capture - buf_idx_end == buf_idx_end_prev
1744474612999: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474613000: capture - frame_sync == 0
1744474613010: capture - buf_idx_end == buf_idx_end_prev
1744474613020: capture - buf_idx_end == buf_idx_end_prev
1744474613030: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474613030: capture - frame_sync == 0
1744474613040: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474613045: capture - frame_sync == 0
1744474613055: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1744474613055: capture - frame_sync == 0
1744474613065: capture - buf_idx_end == buf_idx_end_prev
1744474613076: capture - buf_idx_end != input_buffer.buffer + input_buffer.offset + i
1744474613077: capture - buf_idx_end != input_buffer.buffer + input_buffer.offset + i
....
1744474618638: capture - buf_idx_end != input_buffer.buffer + input_buffer.offset + i
Terminated
```

roleoroleo, 2025-04-12:
Mmmmmh, the buffer is not properly read.
Please send me a copy of the file /dev/shm/fshare_frame_buf

olanwe, 2025-04-13:
@sergioruizsan - just in case the model it is a YRS.3521 - #757 might be helpful. I had exactly the same issue (no connection possible and similar debug output), and this fix solved it for me.

sergioruizsan, 2025-04-20:
Hi @roleoroleo , please find attached a copy of the /dev/shm/fshare_frame_buf
[fshare_frame_buf.zip](https://github.com/user-attachments/files/19826055/fshare_frame_buf.zip)

sergioruizsan, 2025-04-20:
@olanwe - Thanks! Actually with the beta firmware mentioned in this [comment](https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/757#discussioncomment-11652706)  from @roleoroleo it dit work! Now I can see the streaming. Not PTZ as you mention as well though. 

And yes, that's my exact same model as well, Yi Dome Guard YRS.3521, with the ethernet port

roleoroleo, 2025-04-23:
The framebuffer is ok.
What the problem about PTZ?

olanwe, 2025-04-25:
PTZ is working now, see https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/757#discussioncomment-12951469

vgisbert, 2025-05-05:
I want to thank @roleoroleo and the other contributors for their work maintaining this repository.
My camera is a YI Dome Guard YRS.3521 (SN: RFUSY951S4) with Base Version: 12.1.47.05_202411081421

After ascking this repository, I started using the r35gb firmware. Symptoms:
- No image or audio is output via RTSP.
- No image is output via the Snapshot link.
- No images are output in the Snapshot menu or the PTZ control in the web interface.
- The PTZ controls in the web interface do work.

I'm now testing with the r37gb firmware. Symptoms:
- The image in the high-resolution Snapshot link is low-resolution.
- The PTZ controls in the web interface don't work (message: "PTZ functions are not available for this camera model").

I don't know if this will give any clue as to what the appropriate configuration might be for this camera.

roleoroleo, 2025-05-07:
Since the support is in beta, try the version I posted here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/974#issuecomment-2851948438

PTZ problem should be fixed.
About snapshot, could you explain me?

vgisbert, 2025-05-07:
> Since the support is in beta, try the version I posted here: [#974 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/974#issuecomment-2851948438)
> 
> PTZ problem should be fixed. About snapshot, could you explain me?

- PTZ now works. But I got the message once after upgrading to r37gb.
- SNAPSHOT It doesn't work in high resolution (the camera restarts both in the menu and when requesting via HTTP). Since I forced LOW resolution in the menu, it does work (logically).