roleoroleo/yi-hack-Allwinner-v2 #1018: Yi Pro 2K Home Constaly reboot with go2rtc

setterlee, 2025-05-23:
I Have 3 Yi Pro 2K Home working good since 8 months ago without problems, at the moment that the version with go2rtc was realease I installed and was working well till now that one camera started to reboot constanly...

I need some directions to check what can cause this issue. 

With the RTSP disable work well but I need the go2rtc for frigate.

setterlee, 2025-05-23:
I think that I found the issue... Is the camera itself... I reset it and now I cant connect it to the wifi because the camera can't scan the QR code... I assume that the camera is damaged.. .

roleoroleo, 2025-05-24:
Probably there is a memory problem, go2rtc uses a lot of memory.
Remove the sd card and try to start the cam.

RickyTR, 2025-05-27:
I have exactly the same problem. With go2rtc enabled the cam restarts every couple of hours. Everything else is disabled (mqtt, snapshot etc.) and about 3500kb free after reboot. Is there something that I can do to reduce the mem utilization? I'm using the same hack with a DomeU and outdoor 1080 without a problem...

setterlee, 2025-05-27:
> Probably there is a memory problem, go2rtc uses a lot of memory. Remove the sd card and try to start the cam.

My problem now is that I reset the camera and can't scan the QR code to set the wifi SSID. 

My camera stay saying "waiting to connect" even placing the QR code in any position.

I think that the hadware passed away

RickyTR, 2025-05-28:
@setterlee you have 2 Pro 2k Home working with go2rtc without reboot?

setterlee, 2025-05-28:
> @setterlee can I ask you the firmware version? I'm struggling to keep my cam up and running and I would like to understand if it's related to firmware, SD or something else...

Im using the 0.3.3 version

setterlee, 2025-05-28:
> [@setterlee](https://github.com/setterlee) sorry I mean the base version...

You mean, the original cameras firmware? Where can I see that???

BTW, I remenber that I had issues with the web admin interface, so I edited the config file direct via ssh to activate the go2rtc.

setterlee, 2025-05-28:
@RickyTR  here...

[image]