roleoroleo/yi-hack-Allwinner-v2 #1020: Yi Dome Guard - YRS.3521 - RTSP broken on yi-hack-Allwinner-v2 0.3.3 - r35gb

Siavontech, 2025-05-26:
Hi all,

I'm new here. If this isn't the correct way to do it, I apologize.

I followed the entire procedure to load the r35gb firmware on my Yi dome protector, model YRS.3521, and everything works correctly except for the image transmission.

Streaming via the Yi hack homepage link and the snapshot tab doesn't work.

I've already checked port 554 and it's open and working
I activated the credentials for rstp and when I open the links in VLC it asks me for the credentials, so assume the transmission it's working only is missing image.

I took a look and my problem is identical to ticket #909. But with a different camera.

Thank you in advance for your help!

Siavontech, 2025-05-29:
Hi,

I followed the table.
Model YRS.3521
Firmware: 12.1.47.05_202411081421.

roleoroleo, 2025-05-31:
Does it work with the original app?
Could you try ffmpeg client?

Siavontech, 2025-06-01:
Hi,
I tried and got this back.
[Tcp @ 0x6000033759e0] connection refused
[in#0 @ 0x600003d74700] Error opening input: Connection refused

Once again, thank you very much for your help and I apologize if these questions are very silly and simples for you.

roleoroleo, 2025-06-02:
The problem is not simple.
Please check if the buffer is started. Log in with a ssh session and run this command 3 times:
`hexdump -C -n 32 /dev/shm/fshare_frame_buf`

roleoroleo, 2025-06-07:
The buffer is ok.
Try to run rRTPServer in debug mode and post the output:
```
killall wd.sh
killall wd_rtsp.sh
rRTSPServer -m r35gb -r both -a aac -p 554 -d 15
```

roberto1984, 2025-06-29:
Hi,

i just raised an issue about this camera and firmware version, i didn't notice that same issue already reported here.
I'm in identical situation,same camera model and firmware, same issues, i think we can close the one i raised and continue investigating on this.
This is the issue can be closed
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/1030

Thanks in advance, please let us know in case you need further action on our side in order to understand what is happening.

roleoroleo, 2025-07-06:
> ~# killall wd.sh killall: wd.sh: no process killed ~# killall wd_rtsp.sh killall: wd_rtsp.sh: no process killed ~# rRTSPServer -m r35gb -r both -a aac -p 554 -d 15 1749495050860: the size of the buffer is 1048944 1749495050867: capture - mapping file fshare_frame_buf, size 1048944, to b6cb3000 1749495050869: capture - closing the file fshare_frame_buf 1749495050870: capture - frame header size = 26 1749495050871: capture - starting capture main loop 1749495050872: capture - buf_idx_end != input_buffer.buffer + input_buffer.offset + I
> 
> And then it goes into an infinite loop.

Sorry for the delay in the respose.
I need a big log...
1 - Start the process.
2 - Run a rtsp client
3 - Wait 10 seconds
4 - Stop the client
5 - Break the process
6 - Send me the log

Siavontech, 2025-07-07:
Hi,
See the log attached.

[CamLog.txt](https://github.com/user-attachments/files/21107707/CamLog.txt)

roleoroleo, 2025-07-18:
Now I understand your previous comment.
This log is completely wrong.
Please, send me a copy of the file /dev/shm/fshare_frame_buf

James4049, 2025-07-29:
Same issue here also and same camera. Any update on this please? RTSP is not working.

James4049, 2025-07-30:
Thank you, may i ask how do i install this please?

roleoroleo, 2025-07-31:
You have to remove the sd card from the cam and decompress the file in /yi-hack/bin (inside the card).
Rename old file from /yi-hack/bin/rRTSPServer to /yi-hack/bin/rRTSPServer.old

roleoroleo, 2025-08-14:
Strange, this method already works for r30gb.
Any change in the log?

Siavontech, 2025-09-20:
Hi,
check if is right now.

[fshare_frame_buf_dump.txt](https://github.com/user-attachments/files/22444460/fshare_frame_buf_dump.txt)

[rtsp_debug.log](https://github.com/user-attachments/files/22444478/rtsp_debug.log)

roberto1984, 2025-11-16:
Hi @Siavontech did you had the opportunity to try the last file provided by roleoroleo?
i was checking my camera, i think i'm in a slightly different situation, i noticed that SN prefix in my case is RFUS and i'm on firmware 12.1.47 with YI Dome guard camera model YRS.3521, i think this is not supported looking at the supported camera section
@roleoroleo could you please confirm?

roleoroleo, 2025-11-16:
Send me a backup of the flash, I will check it.
Probably, it's already supported by r35gb.

closed 2025-11-19 by Siavontech