roleoroleo/yi-hack-Allwinner-v2 #1030: No snapshots,ssh,ftp.rtsp stream

roberto1984, 2025-06-28:
Hi,

i have an yi yrs.3521, both flashed with firmware 0.3.3 and also 0.3.2
Falsh and wifi connection works ok, i can reach the web page of roleroleo firmware, but:
-Snapshot on the url proposed (http://192.168.1.15/cgi-bin/snapshot.sh?res=low&watermark=yes) are empty, i mean no images
-No RTSP stream
-When i try in ssh it freezes and no prompt

Why is this?lookin in the readme this camera (yi dome guard yi yrs.3521) seems correctly supported with r35gb firmware.

Thanks for any helps.

TungstenE2, 2025-07-08:
Hi all, I ahave similar issue. 

Got 2 new YI Cams, updated FW to 12.0.51.08_202411131107 and Snapshot is empty, but RTSP is working.

2 older YI Cams with FW 12.0.35.8_202211241029 snapshot is still working.

All cams are y291ga.

maatx, 2025-07-22:
I have same issue, RTSP is working but no snapshot.
FW: 0.3.3
Base Version: 12.0.51.04_202311171457
model: y623