roleoroleo/yi-hack-Allwinner-v2 #1050: WIFI issue after rebooting

freegink, 2025-09-21:
This is related #1002. Basically the first 2 bytes of the SSID somehow got messed up. When the camera is rebooted, it won't be able to connect to WIFI. 

One solution I'm thinking is to always call configure_wifi.sh when configure_wifi.cfg exists when the camera is booting up. There's really no harm to call it with every reboot since it checks whether the WIFI config is different.

pedrosimao, 2026-01-04:
I have the same issue.
When turning on the camera after a period without usage, it won't remember the WiFi.
Did you manage to solve the issue?

pedrosimao, 2026-01-04:
Ok, I did some tests and I think I managed to figure out the culprit.
I am using the firmware 0.3.1 on a y211ga.

When I disable the "Enable/disable time OSD" option, the problem no longer happens.

I have tried to update to the 0.3.4 version, but this always bricks my cameras.
For some reason 0.3.1 is more stable.
Hope this helps.

closed 2026-05-06 by github-actions[bot]